wpilibsuite/WPILibPi #35: C++ example doesn't compile, crashes

markkassab, 2019-01-07:
There are a few issues I encountered trying to compile and run the C++ example on the Raspberry:

1. You need to add -lpthread to the Makefile otherwise you can't link.

```
<       ${CXX} -o $@ $< -L/usr/local/frc/lib -lcameraserver -lcscore -lntcore -lwpiutil                                ---
>       ${CXX} -o $@ $< -L/usr/local/frc/lib -lcameraserver -lcscore -lntcore -lwpiutil -lpthread
```

2. main() crashes with an exception because the thread created is not joined.  To fix this, first name the thread object when you create it, then join it.

```
<     std::thread([&] {
---
>     std::thread t1([&] {
```

`>     t1.join();`

3. (Optional) When running on RPi, frc.json need to come from /boot which means __RASPBIAN__ needs to be defined in the Makefile.  

```
<       ${CXX} -O -c -o $@ -I/usr/local/frc/include $<
---
>       ${CXX} -O -c -D__RASPBIAN__ -o $@ -I/usr/local/frc/include $<
```

markkassab, 2019-01-08:
Might make more sense to call t1.detach() rather than t1.join().

PeterJohnson, 2019-01-08:
Yep, that's what I did in my PR.  You can't join there because there can be multiple cameras.

PR #41 (2019-01-10): Fix the C++ example
- Build with -pthread
- Detach the thread to avoid std::thread exception
- Remove `__RASPBIAN__` check (it's not defined by the on-Pi compiler)

Fixes #35.
closed 2019-01-10 by PeterJohnson