wpilibsuite/WPILibPi #45: Streaming to Shuffleboard CameraServer?

rtrahms, 2019-01-08:
I have FRCVision-pi-gen working with two cameras, and can stream the default video streams on a browser on our driver station (frcvision.local:1181 and frcvision.local:1182) with no problem.  I am using the default application setting of streaming multiple cameras.  Bringing up the 2019 release of Shuffleboard, I would expect a CameraServer entry to be available in the side panel, but don't see one.

One other clue: streaming starts on either camera (lifecam blue dot on) when I bring up the respective address above in a browser, but stops (lifecam blue dot off) when I close the browser panel.  This indicates to me that streaming is not always on.  FYI, In an older version of our RPi vision (home brew Raspbian/cscore/opencv RPi Mjpegserver) we would stream continuously from RPi boot.

How do I keep streaming on, or is this a different problem with Shuffleboard CameraServer?

Thanks,
Rob

PeterJohnson, 2019-01-08:
The CameraServer entires in Shuffleboard depend on NetworkTables being connected between Shuffleboard and the Pi.  Do you have a roborio on the same network, and the team number set correctly on the Pi?  Or if you don’t have a Rio on the network, you can set the Pi to NT server mode and set Shuffleboard NT to point directly to the Pi.

The cameras are currently set up to auto-disconnect with no clients, but I plan on changing that for the next release (see #40).

PeterJohnson, 2019-01-09:
Note the NT client tries the 10.TE.AM.2 fixed IP for the roborio as well, so if you’re having mDNS issues I would recommend just changing the roborio to that static ip.

rtrahms, 2019-01-09:
Success! 
All we had to do was start the driver station app, and the camera server instances showed up!
[image]
Thanks for the help - can close this issue.
Rob

PeterJohnson, 2019-01-09:
Most likely the DS app started a mdns responder?  Glad you got it working!

closed 2019-01-09 by PeterJohnson