wpilibsuite/WPILibPi #46: Uploaded python files not printing to web dashboard console

mjansen4857, 2019-01-09:
When I upload a python file to run on the pi, it is run as a bash script instead of through python. This causes a few problems for me. After doing some research, I found that I must add `#!/usr/bin/env python3` to the top of the file. But after doing so I get this error:

`/usr/bin/env: 'python3\r': No such file or directory`

 So I made another script to remove all `\r` characters from the file before I upload it. I either have to do this or edit the runCamera script by replacing `exec ./uploaded.py` with `python3 ./uploaded.py` every time I upload a python file to get it to run. Once I get the file to run, it is shown as running in the Vision Status tab but I receive no output to the console. If I run the python file or the runCamera script manually on the pi I can see the console output, so it is not a problem with the file.

mjansen4857, 2019-01-09:
I should also mention that I get console output for included examples and uploaded Java jars. So I have only experienced this problem with uploaded python files.

mjansen4857, 2019-01-10:
Before I fix the line endings I keep getting the error I mentioned before every five seconds. After fixing the line endings I get one `Waiting 5 seconds...` print then nothing else. The python file I've been trying only prints a string every few seconds as a test. I've tried a couple different files, and I have experienced this same issue in beta 2 as well as the kickoff release, both on a fresh install.

PeterJohnson, 2019-01-11:
Figured it out.  Python output is buffered by default.  I'll set an environment variable in the runCamera script to force Python to unbuffered mode.

mjansen4857, 2019-01-11:
Awesome. This will help a lot. Thank you!

PR #48 (2019-01-11): Convert windows EOL to unix EOL in Python uploads
Fixes #46.
PR #52 (2019-01-11): Default Python output to unbuffered
Fixes #46.
closed 2019-01-11 by PeterJohnson