wpilibsuite/WPILibPi #65: Suggestion: add ATLAS/OpenBLAS to image

auscompgeek, 2019-01-13:
ATLAS and/or OpenBLAS are optimised implementations of BLAS/LAPACK. They will likely give a reasonable speedup for both OpenCV and numpy.

PeterJohnson, 2019-01-14:
It's one or the other, not both.  I have the OpenBLAS package install in #66, but compiling OpenCV with it is going to be more difficult.

auscompgeek, 2019-01-14:
Yeah, I'm not entirely sure which would be better to use, hence why I listed both.

From what I understand, Debian's ATLAS and OpenBLAS packages also provide libblas.so.3 and liblapack.so.3 via their alternatives mechanism, so I believe they'll work as a drop-in replacement without having to specifically link to them.

PeterJohnson, 2019-01-14:
Yes, that works for numpy, but we build our OpenCV, so I have to figure out how to get it to find it when cross-building, etc.

auscompgeek, 2019-01-14:
Ah, right.

Taking a closer look, I found that OpenCV is actually built without LAPACK:

https://github.com/wpilibsuite/FRCVision-pi-gen/blob/frcvision/deps/03-build.sh#L39

From what I can tell, if you pull in the liblapack-dev package (and flip that ON), then it should be fine. Debian builds their OpenCV packages on sid like that it seems.

PR #83 (2019-02-03): Revamp image to build dependencies as part of stages
Fixes #17.

Stage 2 is fairly minimal, stage 3 builds/installs OpenCV and WPILib et al, and stage 4 builds/installs the FRCVision webdash and adds the vision examples.

Other changes:
- OpenCV compiled with ffmpeg, OpenBLAS, and libgtk (fixes #79, fixes #80)
- OpenBLAS added to image (fixes #65)
- C++ Makefile is more easily extensible (fixes #71)
- Sources for everything are bundled into image into /usr/src
- README updated (fixes #16)
- pkg-config files for wpilibc et al are now installed and C++ Makefile uses them (if compiled local to Pi [... cut]
commit c85fd9f3 (2019-02-03): Revamp image to build dependencies as part of stages (#83)

Fixes #17.

Stage 2 is fairly minimal, stage 3 builds/installs OpenCV and WPILib et al, and stage 4
builds/installs the FRCVision webdash and adds the vision examples.

Other changes:
- OpenCV compiled with ffmpeg, OpenBLAS, and libgtk (fixes #79, fixes #80)
- OpenBLAS added to image (fixes #65)
- C++ Makefile is more easily extensible (fixes #71)
- Sources for everything are bundled into image into /usr/src
- README updated (fixes #16)
- pkg-config files for wpilibc et al are now installed and C++ Makefile uses them (if compiled loca [... cut]
closed 2019-02-03 by PeterJohnson