wpilibsuite/WPILibPi #75: cpp-multiCameraServer won't build, won't run on Raspberry Pi

hillegass, 2019-01-16:
After flashing FRC-2019 image to my Raspberry Pi,  I unzipped the cpp-multiCameraServer source. It failed to build until I added the pthreads library to the link line:

```
${EXE}: main.o
	${CXX} -o $@ $< -L/usr/local/frc/lib -lcameraserver -lcscore -lntcore -lwpiutil -lpthread
```

Then, it needed a /home/pi/frc.json file, so I copied the one from /boot/ and edited to put our team number in.

Then when I try to run it, it terminates immediately:
```
./runInteractive 
Waiting 5 seconds...
Setting up NetworkTables client for team 4026
Starting camera 'rPi Camera 0' on /dev/video0
CS: rPi Camera 0: Connecting to USB camera on /dev/video0
CS: rPi Camera 0: SetConfigJson: setting video mode to pixelFormat 1, width 1280, height 720, fps 30
CS: rPi Camera 0: set format 1 res 160x120
CS: rPi Camera 0: Connecting to USB camera on /dev/video0
CS: rPi Camera 0: set format 1 res 1280x720
CS: rPi Camera 0: set FPS to 30
terminate called without an active exception
Aborted
```

Help?

PeterJohnson, 2019-01-16:
Is this with 2019.1.1 or the just released 2019.2.1?  This issue was fixed in 2019.2.1.

PeterJohnson, 2019-02-03:
Closing due to no response; please reply if 2019.2.1 did not fix this issue for you.

closed 2019-02-03 by PeterJohnson