wpilibsuite/WPILibPi #87: ./gradlew build fails because it needs to download Gradle5.0

liloheinrich, 2019-02-03:
Hi, 
I'm from the SOTAbots, the team that asked you to add wpilibj into the frc image, so thanks for adding it to the new image version!

But after setting up the microsd with version 2.1, the Gradle build is failing. So we made changes in examples/java-multiCameraServer/src/main/java then went back to examples/java-multiCameraServer and ran ./gradlew build. It tries to download Gradle from the internet, which it can't access as the image doesn't support using a wireless adapter. 

I hope you can help us figure out what's wrong. I know it may be more of a Gradle problem than a frc image problem but it did pop up in the new image. Here's the error:

pi@frcvision(rw):~/examples/java-multiCameraServer$ ./gradlew build
Downloading https://services.gradle.org/distributions/gradle-5.0-bin.zip

Exception in thread "main" java.net.UnknownHostException: services.gradle.org
        at java.base/java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:220)
        at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:403)
        at java.base/java.net.Socket.connect(Socket.java:591)
        at java.base/sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:285)
        at java.base/sun.security.ssl.BaseSSLSocketImpl.connect(BaseSSLSocketImpl.java:173)
        at java.base/sun.net.NetworkClient.doConnect(NetworkClient.java:182)
        at java.base/sun.net.www.http.HttpClient.openServer(HttpClient.java:474)
        at java.base/sun.net.www.http.HttpClient.openServer(HttpClient.java:569)
        at java.base/sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:265)
        at java.base/sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:372)
        at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1075)
        at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1581)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1509)
        at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:245)
        at org.gradle.wrapper.Download.downloadInternal(Download.java:66)
        at org.gradle.wrapper.Download.download(Download.java:51)
        at org.gradle.wrapper.Install$1.call(Install.java:62)
        at org.gradle.wrapper.Install$1.call(Install.java:48)
        at org.gradle.wrapper.ExclusiveFileAccessManager.access(ExclusiveFileAccessManager.java:69)
        at org.gradle.wrapper.Install.createDist(Install.java:48)
        at org.gradle.wrapper.WrapperExecutor.execute(WrapperExecutor.java:107)
        at org.gradle.wrapper.GradleWrapperMain.main(GradleWrapperMain.java:61)

PeterJohnson, 2019-02-04:
My recommendation (particularly if you can’t get internet access to the Pi via Ethernet) is to build it on desktop and upload the resulting jar via the webdash.  You can download a zip file of the example from the webdash application page, build it on desktop, and upload back to the Pi via the application page, all without building it on the Pi.

liloheinrich, 2019-02-04:
Thanks for your quick reply! I'll try it.

closed 2019-02-04 by liloheinrich