wpilibsuite/WPILibPi #88: OpenCV - Enabling gstreamer

Sn3akyP3t3, 2019-02-04:
Was there any major reason for disabling gstreamer and/or ffmpeg support from the OpenCV library?

As I understand it the gstreamer documentation for OpenCV is cryptic, but appears to be supported.  This may help some teams with a need to stream to their dashboard, but not desiring to package things up within network tables.

I see these flags were the cause:
-DWITH_GSTREAMER=OFF \

These packages will be necessary if going to enable gstreamer support:
sudo apt-get -y install libgstreamer1.0-dev
sudo apt-get -y install libgstreamer-plugins-base1.0-dev

PeterJohnson, 2019-02-04:
I started with the same flags used for the RoboRIO OpenCV build, and the build infrastructure we started with made it difficult to add support for 3rd party libraries like ffmpeg and gstreamer.  That being said, I recently revamped the build infrastructure and along with that enabled ffmpeg support, and it should be relatively easy to add gstreamer support as well.

I'm a little confused by what you mean by "need to stream to their dashboard but not desiring to package things up within networktables".  Streaming doesn't require networktables, networktables just provides the discovery mechanism for the dashboard to find the stream.

PeterJohnson, 2019-02-04:
The image stream is completely independent of NetworkTables; it's delivered as MJPEG over HTTP.

NetworkTables is not used for image data at all; it simply provides a distributed hash table of keys and typed values (e.g. numbers, strings, etc).  NetworkTables was created by WPI, not NI (NI just built the LabView implementation).  At this point NetworkTables has its own ecosystem of applications, and has well understood behavior and interaction with the robot WiFi environment along with controlled latency.  Switching to a different protocol such as MQTT would require significant effort and experimentation, and it's unclear how it would provide a significant benefit to teams over continuing to use NetworkTables.

Sn3akyP3t3, 2019-02-04:
Now I understand what you meant by using NetworkTables to announce the URL of the stream.  I didn't understand that before, but that makes sense.  I've still read a handful of forum posts and conversations where teams were attempting to deliver image data via NetworkTables so that didn't help since I don't have experience with them.

Perhaps this cartoon has some meaning to the situation of selecting an existing protocol:  https://deepstreamhub.com/blog/an-overview-of-realtime-protocols/

My recommendation for MQTT stems from my experience to date with home automation.  It seems to crop up a lot in IOT conversations and tools.  I believe in the idea of pave the cowpaths which seems to be how NetworkTables came to be.  I didn't realize other protocols were analyzed and not found to be suitable.  Newer protocols appear to be showing up that may be more suitable for a second review.  For example, I didn't know LWM2M was a thing.

PR #105 (2019-02-16): Install gstreamer
This also builds OpenCV with gstreamer support.

Fixes #88.
closed 2019-02-16 by PeterJohnson