wpilibsuite/WPILibPi #90: Add Export Config Json Option

Sn3akyP3t3, 2019-02-05:
Its pretty easy once its understood that the default json file created is at /boot/frc.json, but for those that don't want to fiddle with SFTP transfer it would be nice if there was an export option as well as an import option.  The json provided in the example code is somewhat malformed so it didn't pass the import validation when customized.  This export would make it easier to backup settings too.

Perhaps in addition maybe a template can be downloaded with a link to the verbiage for what the setting(s) do.

PeterJohnson, 2019-02-05:
Why are you manually editing it?  Every setting in there is editable via the webdash.

Sn3akyP3t3, 2019-02-05:
Not manually editing it.  Correct, export only for ease of restore or transfer.