wpilibsuite/WPILibPi #91: Vision Application Configuration - Application Set To Custom - Does Not Persist On Reboot

Sn3akyP3t3, 2019-02-05:
[image]
After saving and rebooting I found I wasn't able to start my application from terminal because the camera was already in use.  After a bit of troubleshooting I found another process was running and it was the last uploaded Python application via the Vision Application Configuration screen.  The setting had been undesirably reverted after reboot.

I failed to allocate time to try to recreate this a 2nd time, but I'll be sure to do so tonight.

PeterJohnson, 2019-02-05:
I think this may have been fixed in 2019.2.1.  Please try that version if you haven’t yet.

Sn3akyP3t3, 2019-02-06:
That is the version I'm currently on.  I'll try again soon to confirm the behavior.

PeterJohnson, 2019-02-06:
Oh.  I reproduced it.  The problem is that you can't really "set" custom.  Custom is if the runCamera script has been edited remotely to remove the "TYPE" line.  I'm not sure how to better document this.

Sn3akyP3t3, 2019-02-06:
What would be the advised intermediate solution if someone did this and needs to recover?

I was thinking that upload an empty file and then make sure not to upload anything afterwards should nullify the problem until a fix can arrive.

I know my team will need this to persist.  Our process has classes and multiple files so we can't use the uploader since it works only with single files.  We will have to leverage cron to kick off the process on boot.

PeterJohnson, 2019-02-06:
The webdash just shows what's in the TYPE comment of the runCamera script.  If you edit the runCamera script, just remove the TYPE line, and the webdash will show that as "custom" and never edit the file.

Sn3akyP3t3, 2019-02-06:
K, that did the trick.