wpilibsuite/WPILibPi #92: Line Endings - Need Correction On Import

Sn3akyP3t3, 2019-02-05:
Code files passed over from client to the Pi via the interface should get their line endings corrected prior o execution.  I suggest running dos2unix against uploads.  Probably should do the same for the json import process if not done already.

ThadHouse, 2019-02-05:
Are you using the 2019.2.1 update? This was fixed in that update.

Sn3akyP3t3, 2019-02-06:
I believe so, I downloaded the latest release image the night before opening the ticket.

PeterJohnson, 2019-02-06:
Which application type are you using?  Line endings are only converted for the "Uploaded Python file" application type.

Sn3akyP3t3, 2019-02-06:
Test revealed that line endings are LF and not CRLF after the upload.  I can't recreate so it must be something I did that led to a false alarm.  This one can be closed.

closed 2019-02-07 by AustinShalit