wpilibsuite/WPILibPi #95: Print Build or Version Information On Splash Screen On Bottom Right

Sn3akyP3t3, 2019-02-06:
Just a suggestion for quick confirmation capabilities to the user to be able to view what version of the build they are using.

What alternative method is there to confirm what build or version of the image they are running?

PeterJohnson, 2019-02-06:
Yes, good idea.  Duplicate of #94 (just opened this).

Sn3akyP3t3, 2019-02-06:
Meanwhile what method do people have to confirm what build they are currently running?

I tried a handful of things and this brings me close, but not sure if it is best:
pi@spacialturboencabulator(rw):~$ cat /etc/rpi-issue
Raspberry Pi reference 2019-01-14
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 08163c06810aa506fea56d101e2a1e4d683e0b02, stage2

If that is the build date I'm probably rocking the older kickoff date version, but that may have been me following an old link.

PeterJohnson, 2019-02-06:
That's the build date, so you're running 2019.2.1

PR #110 (2019-02-18): Include git version information in webpage and /etc/rpi-issue
Fixes #95.
commit 3f9cbe8b (2019-02-18): Include git version information in webpage and /etc/rpi-issue (#110)

Fixes #95.
closed 2019-02-18 by PeterJohnson