wpilibsuite/WPILibPi #96: Multiple File Upload Support

Sn3akyP3t3, 2019-02-06:
I'm not sure how complex team's code is, but I'm sure it varies from team to team.  We typically try to work with multiple files whenever we can to keep the code fairly clean.  Unfortunately, this doesn't allow us to use the uploader since it only tolerates single file design.

I propose an enhancement to allow multiple file uploads.  This causes a challenge in that the tool must now have an isolated directory from which it must operate and not just /home/pi because it needs to be able to tidy up between uploads.  It also introduces a need to track which file uploaded is the main().

Another challenge is to block certain file types from being uploaded such as the entire contents of the .git directory if people are using version control.  I'm not sure, but Python cached files might also pose an issue.

Anyway, this is just an idea and sounds reasonable and feasible, but it may not depending on what the capabilities.  I'm confident that if it is a feature added it will get used.

PeterJohnson, 2019-02-06:
I'm a bit on the fence on this.  I think the right solution long term is probably to follow more of a RoboRIO style approach using SSH to deploy with GradleRIO for C++ and Java and a Python uploader for Python.

Sn3akyP3t3, 2019-02-06:
That sounds like a reasonable solution and that is what we currently are doing.

Maybe what I'm asking for doesn't belong anywhere near the current process because that should remain a simple process as-is.  Perhaps something more suited for a more advanced tool to build and deploy.  I can't think of a particular tool that covers C++, Java, and Python... maybe Apache ant?  Not sure really how to fill this gap.