wpilibsuite/WPILibPi #98: Upgrade OpenCV To 4.0.1

Sn3akyP3t3, 2019-02-07:
According to publications OpenCV 4 has many optimizations which should allow the Pi to be more capable of data analysis which could give more teams a competitive edge and extend the life of their existing hardware to reduce strain on their budgets.

Doubtful teams are using much deep learning algorithms, but this also enables a lot more in that realm as well should they care to dabble.

ThadHouse, 2019-02-07:
We can't update to 4 currently because cscore still requires 3. With breaking changes in the headers, supporting both isn't super easy to do, so this definitely would not be something that happens for this year. This summer we will look again, but we will always have to match the cscore requirement.

Sn3akyP3t3, 2019-02-07:
np, maybe belongs on a roadmap then.  Didn't see a roadmap of this project so just issued it as an issue.

closed 2022-02-13 by PeterJohnson