wpilibsuite/WPILibPi #99: Troubles grabbing settings file

ColeScheller, 2019-02-08:
I have the newest available image for frcvision

When I attempt to upload your code I get the following error:

```
Traceback (most recent call last):

  File "./uploaded.py", line 684, in <module>

    if not readConfig():

  File "./uploaded.py", line 631, in readConfig

    j = json.load(f)

  File "/usr/lib/python3.5/json/__init__.py", line 265, in load

    return loads(fp.read(),

  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode

    return codecs.ascii_decode(input, self.errors)[0]

UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 174: ordinal not in range(128)

Waiting 5 seconds...
```
I attempted to comment that section out, to no avail.

This error occurs anytime I try to execute anything via the web dashboard. The pi was re-imaged today, so I doubt it's that.

PeterJohnson, 2019-02-08:
Can you grab the /boot/frc.json file from the Pi and upload it as an attachment here?  That file is a text file and shouldn't have any binary data in it.

ColeScheller, 2019-02-09:
Github won't let me upload a .json, so here is it pasted:

```
{
    "team": 3656,
    "cameras": [
	{
	    "name": "rPi Camera 0",
	    "path": "/dev/video0",
	    "pixel format": "mjpeg",
	    "width": 160,
	    "height": 120,
	    "fps": 30
	}
    ]
}
```

ColeScheller, 2019-02-09:
That is the root of the problem. My issue now is that I can't get the camera without that address.

PeterJohnson, 2019-02-09:
Agreed there's a bug here but the workaround is straightforward.  You can get the camera without that address.  If you only have one camera, it will always be /dev/video0.  The path you've set up is only necessary to disambiguate cameras when you have more than one.

PeterJohnson, 2019-02-09:
Actually there's an easier solution.  See https://stackoverflow.com/questions/35051867/json-load-function-give-strange-unicodedecodeerror-ascii-codec-cant-decod
Add a 3rd parameter to the open() line on line 94 or so, `encoding='utf-8'`.  I'll PR a fix.

ColeScheller, 2019-02-09:
That fixed it and it works perfectly now, I'm going to close it.

PR #101 (2019-02-10): Python example: Use UTF-8 encoding when opening frc.json
Fixes #99.
closed 2019-02-09 by ColeScheller