wpilibsuite/WPILibPi #106: Getting this message when trying to build with the example Java project

GhostAction, 2019-02-13:
[PS C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer> ./gradlew build

FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\build.gradle' line: 8

* What went wrong:
A problem occurred evaluating root project 'java-multiCameraServer'.
> Plugin with id 'com.github.johnrengelman.shadow' not found.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full
insights.

* Get more help at https://help.gradle.org](url)

PeterJohnson, 2019-02-13:
Are you building with Java 11?  Are you connected to the internet?

GhostAction, 2019-02-17:
That was it thank you.

closed 2019-02-17 by GhostAction