wpilibsuite/WPILibPi #107: The local URL doesn't load

TheHardestBunnyAlive, 2019-02-14:
Hi,

I have just flashed my SD card with the image and when I went to go to http://frcvision.local/ the page tells me that it can't be reached.

The Raspberry Pi is turned on and plugged into my computer with an ethernet cable.

Any help is much appreciated.

PeterJohnson, 2019-02-14:
There are usually two potential causes for this:
1) you don't have a mDNS responder installed such as Bonjour (installed with iTunes) or the NI update suite (which installs a NI mDNS responder)
2) your computer isn't getting a suitable link-local IP address

One way to work around (2) is to plug both the Pi and your computer into a network with a DHCP server (such as on most LANs).  This still needs a mDNS resolver to find the .local address.

A short-term solution is to hook up a HDMI monitor and USB keyboard to the Pi, log in with "pi", "raspberry", and run "ifconfig -a", which will print the IP address of the Pi ethernet.  You can use that IP address to access the Pi rather than "frcvision.local".  You can set a fixed IP address via the web dashboard.

TheHardestBunnyAlive, 2019-02-15:
> There are usually two potential causes for this:
>
> 1. you don't have a mDNS responder installed such as Bonjour (installed with iTunes) or the NI update suite (which installs a NI mDNS responder)
> 2. [...]

So when I go into the raspberry pi, login and type "sudo ifconfig -a", I get the ip "169.254.44.169" which I then put into my browser. I get the same message about the site not being able to be reached.

PeterJohnson, 2019-02-16:
Can you ping that IP address from the computer?  It could be a firewall issue.

ThePfaffanater, 2019-03-26:
I am having this same problem I can ping, and I am using that address that I pinged.

PeterJohnson, 2019-03-28:
Can you ssh in (e.g. using PuTTY)?  What browser are you using?

ThePfaffanater, 2019-03-28:
I figured out the problem with this one. I could SSH into it. The problem appears to have been that I was using a RaspberryPi 2. Putting it on a 3 came up with no issues.

PeterJohnson, 2019-03-28:
Huh, that's odd, it *should* work on a Pi 2, but I admit I've not tested that platform.