wpilibsuite/WPILibPi #108: Trying to integrate GRIP

GhostAction, 2019-02-17:
Hi, I'm trying to get my generated GRIP pipeline running on the Raspberry Pi but I am having issues implementing it.

`VisionThread visionThread = new VisionThread(cameras.get(0),
              new GripPipeline(), pipeline -> {
      });`
It is saying visionThread is undefined. How would I fix this?

Thanks!

PeterJohnson, 2019-02-17:
You will need to import it from edu.wpi.first.wpilibj.vision.VisionThread.

GhostAction, 2019-02-18:
Already have it imported.

PeterJohnson, 2019-02-18:
Please post the exact error message.

GhostAction, 2019-02-18:
`PS C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer> ./gradlew build
Starting a Gradle Daemon, 1 busy and 1 stopped Daemons could not be reused, use --status for details

> Task :compileJava FAILED
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:209: error: class, interface, or enum expected
  public static void main(String... args) {
                ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:212: error: class, interface, or enum expected
    }
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:217: error: class, interface, or enum expected
    }
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:221: error: class, interface, or enum expected
    if (server) {
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:223: error: class, interface, or enum expected
      ntinst.startServer();
      ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:224: error: class, interface, or enum expected
    } else {
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:226: error: class, interface, or enum expected
      ntinst.startClientTeam(team);
      ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:227: error: class, interface, or enum expected
    }
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:231: error: class, interface, or enum expected
    for (CameraConfig cameraConfig : cameraConfigs) {
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:233: error: class, interface, or enum expected
    }
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:244: error: class, interface, or enum expected
      visionThread.start();
      ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:245: error: class, interface, or enum expected
    }
    ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:248: error: class, interface, or enum expected
    for (;;) {
           ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:251: error: class, interface, or enum expected
      } catch (InterruptedException ex) {
      ^
C:\Users\Mattawan WiredCats\Documents\java-multiCameraServer\src\main\java\Main.java:253: error: class, interface, or enum expected
      }
      ^
15 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full
insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 18s
1 actionable task: 1 executed`

PeterJohnson, 2019-02-18:
What comes before line 209 in Main.java?  Seeing the whole file would be ideal.

GhostAction, 2019-02-18:
```/*----------------------------------------------------------------------------*/
/* Copyright (c) 2018 FIRST. All Rights Reserved.                             */
/* Open Source Software - may be modified and shared by FRC teams. The code   */
/* must be accompanied by the FIRST BSD license file in the root directory of */
/* the project.                                                               */
/*----------------------------------------------------------------------------*/

import java.io.IOException;
import java.nio.file.Files;
import java.nio.file.Paths;
import java.util.ArrayList;
import java.util.List;

import com.google.gson.Gson;
import com.google.gson.GsonBuilder;
import com.google.gson.JsonArray;
import com.google.gson.JsonElement;
import com.google.gson.JsonObject;
import com.google.gson.JsonParser;

import edu.wpi.cscore.MjpegServer;
import edu.wpi.cscore.UsbCamera;
import edu.wpi.cscore.VideoSource;
import edu.wpi.first.cameraserver.CameraServer;
import edu.wpi.first.networktables.NetworkTableInstance;
import edu.wpi.first.vision.VisionPipeline;
import edu.wpi.first.vision.VisionThread;

import org.opencv.core.Mat;

/*
   JSON format:
   {
       "team": <team number>,
       "ntmode": <"client" or "server", "client" if unspecified>
       "cameras": [
           {
               "name": <camera name>
               "path": <path, e.g. "/dev/video0">
               "pixel format": <"MJPEG", "YUYV", etc>   // optional
               "width": <video mode width>              // optional
               "height": <video mode height>            // optional
               "fps": <video mode fps>                  // optional
               "brightness": <percentage brightness>    // optional
               "white balance": <"auto", "hold", value> // optional
               "exposure": <"auto", "hold", value>      // optional
               "properties": [                          // optional
                   {
                       "name": <property name>
                       "value": <property value>
                   }
               ],
               "stream": {                              // optional
                   "properties": [
                       {
                           "name": <stream property name>
                           "value": <stream property value>
                       }
                   ]
               }
           }
       ]
   }
 */

public final class Main {
  private static String configFile = "/boot/frc.json";

  @SuppressWarnings("MemberName")
  public static class CameraConfig {
[... 162 lines cut ...]
    }

    // start image processing on camera 0 if present
    if (cameras.size() >= 1) {
      //VisionThread visionThread = new VisionThread(cameras.get(0),
        //      new MyPipeline(), pipeline -> {
        // do something with pipeline results
      //});
      VisionThread visionThread = new VisionThread(cameras.get(0),
              new GripPipeline(), pipeline -> {
      });
      visionThread.start();
    }

    // loop forever
    for (;;) {
      try {
        Thread.sleep(10000);
      } catch (InterruptedException ex) {
        return;
      }
    }
  }
}
```

PeterJohnson, 2019-02-18:
You have an extra close brace after the “example pipeline” comment.  Java is seeing this as closing the class.