wpilibsuite/WPILibPi #118: Streaming multiple of the same camera types

maxwellpettit, 2019-02-19:
We are using multiple Microsoft LifeCam 3000's connected to the Pi and want to make sure that we can identify each camera.  Normally, we configure the camera source as "/dev/video0", "/dev/video1", etc.  

However, if we plug the cameras in different ports or plug them into the Pi in a different order, the output streams will not be the same.  Since they are all the same model of camera, we cannot tell them apart via model/vendor ID.

Is there a way to configure these cameras or assign a camera ID at a firmware level to ensure that we load each camera in the same order? I have not been able to find any information about this.

Is the only solution to this to ensure each camera is plugged in the proper port/order?

PeterJohnson, 2019-02-19:
Use the /dev/v4l/by-path alternate device path (this is available in the 2019.2.1 and later versions via the web dashboard, but you can do it manually in 2019.1.1).  This device path is based on the physical port the camera is plugged into.

maxwellpettit, 2019-02-19:
> This device path is based on the physical port the camera is plugged into

This is what I meant when I asked:
> Is the only solution to this to ensure each camera is plugged in the proper port/order?

So in this case, the answer is no, we have to rely on the port that the camera is plugged into.

Thanks,
Max

closed 2019-02-19 by maxwellpettit