wpilibsuite/WPILibPi #121: Connection/communication with pi fails when on a robot network

mjansen4857, 2019-02-20:
We have been trying to add our vision onto the robot but have been running into a lot of connection issues. When we directly connect Ethernet from out laptop to the pi we can access the web dashboard and get data from it just fine, but once we connect it to our robot the connection becomes really inconsistent. Sometimes it will work fine, but most of the time we either can't access the dashboard or can't view streams and get network tables data, and sometimes we can't do either. We have to restart the pi multiple times until we get lucky and everything works. We've tried plugging the pi into the radio and through a switch but neither seem to change anything. We've tried a few things to find the problem such as letting the robot boot first, then the pi, but nothing we do seems to help. This happens on multiple robots, pis, and laptops. The code were running is the example code with some opencv processing added on top of it.

PeterJohnson, 2019-02-21:
It sounds like you're having issues with mDNS, in particular with the Pi trying to find the Rio via mDNS.  I would recommend switching to static IPs, especially for the roboRio.  This can be done by setting the IP on the roboRio to 10.x.y.2 and setting the Pi to 10.x.y.10 or similar.

mjansen4857, 2019-02-22:
This seemed to work well tonight, thanks for the help!

closed 2019-02-22 by mjansen4857

PeterJohnson, 2019-02-22:
How is your robot network set up?  Are you using a radio with this year’s firmware and set to the same team number as the Rio?

mjansen4857, 2019-02-22:
Everything is recent and it's all set to the same number. We have a Poe cable connecting the radio to the Rio, and we usually connect the other port to a switch for tethering and other devices like the pi, although we did not use it tonight.