wpilibsuite/WPILibPi #123: Camera Settings not updating for exposure

AkshatJain1, 2019-03-11:
Using a Microsoft Lifecam 3000.

I put exposure_auto to manual and move exposure_absolute to 1 and the camera stream reflects these changes. However, once I click 'Copy Source Config' back in the web console and then click 'Save' at the bottom, it does not save the exposure. I click 'Open Camera Stream' and exposure_auto is set to automatic and the camera stream does not reflect my exposure.

I tried directly copying the source config instead of the 'Copy Source Config' button, but in the 'vision status' tab of the console, I am getting the following errors and the camera is still not reflecting the exposure changes.

CS: ERROR: ioctl VIDIOC_S_EXT_CTRLS failed at UsbCameraProperty.cpp:65: Invalid argument (UsbUtil.cpp:156)
CS: ERROR: ioctl VIDIOC_S_EXT_CTRLS failed at UsbCameraProperty.cpp:65: Input/output error (UsbUtil.cpp:156)
CS: ERROR: ioctl VIDIOC_S_EXT_CTRLS failed at UsbCameraProperty.cpp:65: Input/output error (UsbUtil.cpp:156)

PeterJohnson, 2019-03-11:
If all you want to do is set manual exposure, don't use Copy Source Config.  Clear that text box and just set the "exposure" setting to a number.  I have heard some reports of exposure not consistently taking on some cameras, but the Lifecam should work.

AkshatJain1, 2019-03-12:
I did this and clicked save but my settings are not changing when I open the camera stream.

AkshatJain1, 2019-03-12:
This worked. Even though when I click open camera stream, the settings don't change, the console output is setting the exposure to 1 without any errors.

closed 2019-03-12 by AkshatJain1