wpilibsuite/WPILibPi #124: On installing opencv-contrib for Raspi

jessicaanhdao, 2019-03-12:
Hi there,

I'm writing a program for the Roborio and for vision, I'm thinking about using ArucoTag from opencv-contrib package (which is not contained in regular opencv package). Since Roborio doesn't not support the installation for opencv-contrib, I was suggested to move onto the raspberry pi. However, I want to ask if it's feasible to install extra modules onto the raspi.

https://github.com/opencv/opencv_contrib
https://pypi.org/project/opencv-contrib-python/

Thank you!

Sn3akyP3t3, 2019-03-13:
This is a build from the ground up so the links you provided won't be usable for this, but I'm sure its well understood.  I'm interested in expanding the options horizon as well.  Looks like this repo is the one to be referred to https://github.com/opencv/opencv_contrib and also make note of the warning regarding introduction of potential instability.  Seems to be a common trade-off with more features comes great instability...

auscompgeek, 2019-03-13:
You will likely be better off using a different image with a distro-provided OpenCV 3.  I, one of the RobotPy devs, package robotpy-cscore for some different distros here: https://software.opensuse.org/download.html?project=home%3Aauscompgeek%3Arobotpy&package=python3-cscore

The two simplest distros that I package for that will likely be simple to get running on a Raspberry Pi from scratch are probably Raspbian testing and Fedora. From a quick glance, it looks like the Python bindings on both of those distros support the OpenCV contrib bits.

Good luck!