wpilibsuite/WPILibPi #128: Feature Request - Add Pi Camera Library With Array Support

FRC-3277, 2019-03-14:
Missing libraries when asked for in default image.

Please see https://www.pyimagesearch.com/2015/03/30/accessing-the-raspberry-pi-camera-with-opencv-and-python/
for comments about adding the optional array sub-module which is essential for OpenCV

from picamera.array import PiRGBArray
from picamera import PiCamera

ThadHouse, 2019-03-14:
You can actually use the CameraServer stuff already installed to grab images from the Pi Camera, and do OpenCV and MJPEG streaming from that. The Pi Camera works and has been tested with the libraries, and looks just like a USB camera. So following any of the Python USB Camera examples in the image will work for the Pi Camera.

FRC-3277, 2019-03-14:
So just call the picamera the same way one would refer to a USB camera?  I didn't see any reference in code documentation or over in the robotpy documentation mentioning this.  Could both be updated to give an example of this?

So basically just keep using this?
camera = cs.UsbCamera("usbcam", 0)

ThadHouse, 2019-03-14:
Yup. It looks just like a USB camera.

FRC-3277, 2019-03-15:
Getting a lot errors.  Checked again to ensure camera module enabled.  Found nothing wrong there.

I checked picamera wiring and found it was correct.

Ran this and got an error.
raspistill -o image.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

Does that mean I need to do some software install/updates to make it work?

PeterJohnson, 2019-03-15:
Are you trying to run raspistill when the camera server application is running?  That won't work, only one program can access the camera at a time.  You can disable the background application by either running `sudo svc -d /service/camera` at the command line or clicking "Down" on the Vision Status tab of the webdash.

Sn3akyP3t3, 2019-03-15:
Thanks, I'll check this in the morning.  Is there a list of CLI commands and documentation for some of this?  I often forget to check the webdash, but if thats all there is then so be it.  I might just create a bash startup script to force the camera server to keep off in case it somehow gets toggled.

Also, if I modify fstab does that cripple the RO and RW aliases?  Trying to remove the RO behavior and looking to apply it there.

FRC-3277, 2019-03-15:
The errors I was getting was exactly because the camera was already in use.  I forgot about this behavior: https://github.com/wpilibsuite/FRCVision-pi-gen/issues/91 and when I applied the fix the functionality was restored.

FRC-3277, 2019-03-15:
From what I know now this isn't an issue

closed 2019-03-15 by FRC-3277