wpilibsuite/WPILibPi #132: import cv2 fails on python3 on 2020.1.1

kglasnapp, 2019-10-07:
I attempted import cv2  on python3 on 2020.1.1 -- the result was file not found.  

I did an import cv2 with a 2019.3.1 image and it worked. What am I doing wrong?

PeterJohnson, 2019-10-08:
You're not doing anything wrong, looks like a bug in the 2020 beta image.

auscompgeek, 2019-10-08:
It looks like the stage3 build does look for Python 3.7 correctly, but the package list includes python3.5 instead of python3.7.

joecaraccio, 2019-10-10:
Also experiencing this..we were so confused how the image didn't include cv2.

Does anyone know how to generate a release of the image to be flashed? I spend most of my time on a windows machine so didn't want to deal with setting up all that stuff

closed 2019-10-12 by PeterJohnson

PeterJohnson, 2019-10-12:
There's an easy patch fix if you don't want to reflash the whole image, just ssh in, make the filesystem writable with `rw`, and run `sudo ln -sf /usr/local/frc/lib/python3.7/dist-packages/cv2 /usr/local/lib/python3.7/dist-packages/cv2`

PR #134 (2019-10-12): Fix python3 cv2 paths
Fixes #132
PR #133 (2019-10-12): stage3: Grab python3.7 packages
This should fix #132.