wpilibsuite/WPILibPi #141: C++ example does not  make...

mddrouillard, 2019-11-30:
I may be missing a basic step, after booting a fresh image on the pi, finding the camera, saving the camera info so that the json is created, I ssh to the pi  and cd to the 

pi@frcvision(rw):~/examples/cpp-multiCameraServer$
 
directory.

As per the example instructions in README, I issue a simple make, but this is what happens...
as a result I dont get a new executable that can be installed into place with make install.

Any ideas?

pi@frcvision(rw):~/examples/cpp-multiCameraServer$ make
g++ -pthread -g -Og -c -o main.o  -I/usr/local/frc/include -I/usr/local/frc/include/opencv -I/usr/local/frc/include main.cpp
In file included from /usr/local/frc/include/wpi/ArrayRef.h:15,
                 from /usr/local/frc/include/networktables/NetworkTableInstance.h:17,
                 from main.cpp:13:
/usr/local/frc/include/wpi/STLExtras.h:585:13: error: ‘optional’ in namespace ‘std’ does not name a template type
     -> std::optional<typename std::remove_const<
             ^~~~~~~~
/usr/local/frc/include/wpi/STLExtras.h:585:8: note: ‘std::optional’ is only available from C++17 onwards
     -> std::optional<typename std::remove_const<
        ^~~
/usr/local/frc/include/wpi/STLExtras.h:585:21: error: expected initializer before ‘<’ token
     -> std::optional<typename std::remove_const<
                     ^
/usr/local/frc/include/wpi/STLExtras.h:594:12: error: ‘optional’ in namespace ‘std’ does not name a template type
       std::optional<typename std::remove_const<typename std::remove_reference<
            ^~~~~~~~
/usr/local/frc/include/wpi/STLExtras.h:594:7: note: ‘std::optional’ is only available from C++17 onwards
       std::optional<typename std::remove_const<typename std::remove_reference<
       ^~~
In file included from /usr/local/frc/include/networktables/NetworkTableInstance.h:17,
                 from main.cpp:13:
/usr/local/frc/include/wpi/ArrayRef.h:65:44: error: invalid use of ‘::’
     /*implicit*/ ArrayRef(std::nullopt_t) {}
                                            ^
/usr/local/frc/include/wpi/ArrayRef.h:65:44: error: expected ‘;’ at end of member declaration
     /*implicit*/ ArrayRef(std::nullopt_t) {}
                                            ^
                                             ;
/usr/local/frc/include/wpi/ArrayRef.h:296:48: error: function definition does not declare parameters
     /*implicit*/ MutableArrayRef(std::nullopt_t) : ArrayRef<T>() {}
                                                ^
In file included from main.cpp:17:
/usr/local/frc/include/wpi/json.h:250:40: error: ‘conjunction’ is not a member of ‘std’
                                   std::conjunction<std::negation<std::is_same<void, CompatibleObjectType>>,
                                        ^~~~~~~~~~~
/usr/local/frc/include/wpi/json.h:250:40: note: suggested alternative: ‘functio ’
                                   std::conjunction<std::negation<std::is_same<void, CompatibleObjectType>>,
                                        ^~~~~~~~~~~
                                        function
/usr/local/frc/include/wpi/json.h:250:40: error: ‘conjunction’ is not a member of ‘std’
/usr/local/frc/include/wpi/json.h:250:40: note: suggested alternative: ‘functio ’
                                   std::conjunction<std::negation<std::is_same<void, CompatibleObjectType>>,
                                        ^~~~~~~~~~~
                                        function
/usr/local/frc/include/wpi/json.h:250:57: error: ‘negation’ is not a member of  std’
                                   std::conjunction<std::negation<std::is_same<void, CompatibleObjectType>>,
                                                         ^~~~~~~~
/usr/local/frc/include/wpi/json.h:250:57: note: suggested alternative: ‘negate’
                                   std::conjunction<std::negation<std::is_same<void, CompatibleObjectType>>,
                                                         ^~~~~~~~
                                                         negate
/usr/local/frc/include/wpi/json.h:250:105: error: wrong number of template arguments (1, should be 3)
       std::conjunction<std::negation<std::is_same<void, CompatibleObjectType>>,
                                                                             ^~

/usr/local/frc/include/wpi/json.h:236:8: note: provided for ‘template<bool B, class RealType, class CompatibleObjectType> struct wpi::detail::is_compatible_object_type_impl’
 struct is_compatible_object_type_impl : std::false_type {};
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 76 lines cut ...]
         return value(ptr, std::string(default_value));
                                                     ^
/usr/local/frc/include/wpi/json.h:5002:15: note: candidate: ‘template<class ValueType, typename std::enable_if<std::is_convertible<wpi::json, ValueType>::value, int>::type <anonymous> > ValueType wpi::json::value(wpi::StringRef, const ValueType&) const’
     ValueType value(StringRef key, const ValueType& default_value) const
               ^~~~~
/usr/local/frc/include/wpi/json.h:5002:15: note:   template argument deduction/substitution failed:
/usr/local/frc/include/wpi/json.h:5097:22: note:   cannot convert ‘ptr’ (type ‘const json_pointer’ {aka ‘const wpi::json_pointer’}) to type ‘wpi::StringRef’
         return value(ptr, std::string(default_value));
                      ^~~
/usr/local/frc/include/wpi/json.h:5024:17: note: candidate: ‘std::__cxx11::string wpi::json::value(wpi::StringRef, const char*) const’
     std::string value(StringRef key, const char* default_value) const
                 ^~~~~
/usr/local/frc/include/wpi/json.h:5024:17: note:   no known conversion for argument 1 from ‘const json_pointer’ {aka ‘const wpi::json_pointer’} to ‘wpi::StringRef’
/usr/local/frc/include/wpi/json.h:5072:15: note: candidate: ‘template<class ValueType, typename std::enable_if<std::is_convertible<wpi::json, ValueType>::value, int>::type <anonymous> > ValueType wpi::json::value(const json_pointer&, const ValueType&) const’
     ValueType value(const json_pointer& ptr, const ValueType& default_value) const
               ^~~~~
/usr/local/frc/include/wpi/json.h:5072:15: note:   template argument deduction/substitution failed:
/usr/local/frc/include/wpi/json.h:5071:78: error: no type named ‘type’ in ‘struct std::enable_if<false, int>’
                  std::is_convertible<json_t, ValueType>::value, int>::type = 0>
                                                                              ^
/usr/local/frc/include/wpi/json.h:5095:17: note: candidate: ‘std::__cxx11::string wpi::json::value(const json_pointer&, const char*) const’
     std::string value(const json_pointer& ptr, const char* default_value) const
                 ^~~~~
/usr/local/frc/include/wpi/json.h:5095:17: note:   no known conversion for argument 2 from ‘std::__cxx11::string’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
make: *** [Makefile:22: main.o] Error 1

auscompgeek, 2019-12-01:
Well that's awkward. The example Makefile doesn't set `-std=c++17`.

mddrouillard, 2019-12-01:
many thanks.

I added the following to the Makefile (between DEPS_CFLAGS and DEPS_LIBS and it now builds.

DEPS_CFLAGS?=$(shell env PKG_CONFIG_PATH=/usr/local/frc/lib/pkgconfig pkg-config
 --cflags wpilibc)
CXXFLAGS?=-std=c++17****
DEPS_LIBS?=$(shell env PKG_CONFIG_PATH=/usr/local/frc/lib/pkgconfig pkg-config -
-libs wpilibc)
EXE=multiCameraServerExample
DESTDIR?=/home/pi/

Now how to figure out the rest of this.

Many Thanks AuscompGeek!

PR #142 (2019-12-02): Set std=c++17 in C++ example
Fixes #141.
closed 2019-12-02 by PeterJohnson