wpilibsuite/WPILibPi #147: Out of memory error with simple fix

rakar, 2020-01-10:
I noticed that after running for a while (much more than an FRC match) my pipeline code would eventually crash due to an Out of Memory error. I corrected this by adding a System.gc() line to the endless loop at the bottom of Main. As expected the free memory on the webpage drops every second for 10 seconds and then cycles back to the original number and starts dropping again. All good after an overnight run. 

```java
// loop forever
    for (;;) {
      try {
        Thread.sleep(10000);
        System.gc();
      } catch (InterruptedException ex) {
        return;
      }
    }
```

ThadHouse, 2020-01-10:
What is likely happening is you are allocating either Mats or other native data structures during pipeline execution, rather then only before loop execution. The GC doesn't know how to properly handle these native resources, and gets confused. Anything allocated during the loop, if it has a `close` method should have closed called at the end of the loop, or placed in a try with resources. that is a much more reliable and better option then manually forcing the GC to run.

rakar, 2020-01-10:
I do clone a mat which I was deliberately not releasing. My concern was that I didn't know if the mat I was creating would still be in use when I released it (due to an imageSource.putFrame(output) call). I have tried releasing it and it seems to work, but my, limited, understanding is that in general one can't be sure that a mat is no longer in use when passed off to some black box routine in the opencv world. 
The memory usage with the release in place and the gc() gone is wonderfully flat-line. I just want to be sure that this is generally considered safe.

ThadHouse, 2020-01-10:
putFrame does explicitly make an internal copy so it is safe to free the mat after that. Same with something like imshow. The functions that display usually make a copy internally specifically for this reason.