wpilibsuite/WPILibPi #152: Unable to connect to roboradio on startup

Jpiv1212, 2020-01-25:
So I'm using a raspberry pi for vision and I'm having an issue with it when the robot is turned on. The pi doesn't seem to connect to the network and is impossible to reach unless I cycle power while the robot is still on. I think it's because the pi is starting up before the radio is, but I don't know how to fix that other than manually cycling power after robot startup

AustinShalit, 2020-01-25:
This is strange.  To help us debug this issue, can you please try to reflash your radio with the latest firmware? The latest version of the utility is 2020.0.0. Instructions can be found here: https://docs.wpilib.org/en/latest/docs/getting-started/getting-started-frc-control-system/radio-programming.html

Once you have completed that, please attempt your test again.

Jpiv1212, 2020-01-25:
After flashing to the latest firmware (It did happen to be old) the issue hasn't gone away. I still had to manually cycle power on the rPi after turning the robot on. I did happen to notice another issue that while after doing that I was able to connect and stream camera data, for some reason the rPi wasn't pushing data to networktables, which was fixed by restarting the code on the pi

PeterJohnson, 2020-01-26:
Sometimes we've seen connectivity issues between the radio ports.  You may get more reliable behavior by adding a small network switch to your robot and connecting the Pi, Rio, and radio to it.