wpilibsuite/WPILibPi #171: Basic Vision Example missing a dependency

williamhaley, 2020-12-14:
The [Basic Vision Example](https://docs.wpilib.org/en/latest/docs/software/vision-processing/frcvision/basic-vision-example.html) ([src](https://github.com/wpilibsuite/frc-docs/blob/master/source/docs/software/vision-processing/frcvision/basic-vision-example.rst)) code does not seem to work with the `WPILibPi_image-v2021.1.1-beta-4` release.

I know this work is still flagged as in progress/unstable, but wanted to raise the issue in case it's something that was overlooked.

Not sure if this is an issue with the documentation or the image (or me :-D, if I'm overlooking a step)

```
Waiting 5 seconds...

Traceback (most recent call last):
  File "uploaded.py", line 1, in <module>

    from cscore import CameraServer
  File "/usr/local/lib/python3.7/dist-packages/cscore/__init__.py", line 1, in <module>
    from _cscore import *
```

PeterJohnson, 2020-12-14:
This is likely an issue with the image.  I'll look into it.

williamhaley, 2020-12-14:
Thanks @PeterJohnson! I confirmed that I can get past that issue using the stable `FRCVision-v2020.2.1_2020-01-19` image.

I think the docs for the example do also need some updates though. It looks like `main` isn't invoked, `sink` is undefined, `time` needs to be imported, and `json.load` needs a file reference like... `json.load(open('/boot/frc.json'))`.

I'll see if I can get the example code running on the stable image and open a PR for that code, but I may be out of my depth and hit a wall.

PR #172 (2021-01-02): Fix robotpy-cscore install
Also upgrade to robotpy-cscore 2020.0.1.

Partially fixes #171 (ntcore is still missing).
closed 2021-01-02 by PeterJohnson