wpilibsuite/WPILibPi #191: Updating Network Settings

k4rstm4n, 2021-01-24:
I am reconfiguring the network settings so that the Pi uses a static IP instead of the default DHCP. When I set it to writable, change the settings, and hit save on the Network Settings tab, it doesn’t save. If I close the web dashboard and open it back up again, the changes are gone. I can save things just fine on other tabs, and I’ve been able to make this change before in previous versions of the image. Could I be doing something wrong or is this a bug?

PeterJohnson, 2021-01-24:
I assume this is on the base image (not the Romi image) since the default is DHCP?

k4rstm4n, 2021-01-24:
Yep. Also, sorry for putting this in the wrong place.

PeterJohnson, 2021-01-24:
Most likely something got broken with all the Romi upgrades.  I'll take a look.

auscompgeek, 2021-01-24:
I was looking into this earlier - the web config does correctly read the dhcpcd.conf when I added the settings manually (with the comment it expects) at least.

PR #193 (2021-01-28): Fix network settings for non-Romi
Fixes #191.
closed 2021-01-28 by PeterJohnson