wpilibsuite/WPILibPi #195: Connection issiue with 2021.2.1

julian-steiner, 2021-02-04:
We can't connect to our raspberry pi model 3 b+ over the link http://wpilibpi.local/ and with the ip address of the raspberry pi.
But we can ping it and we can ssh into it without any further problems. 

When we show the running servers with netstat -tulpn on the pi connected over ssh, we see that the http port is not running, so we can't connect to it on the website

also the folder /usr/local/frc is missing on the pi

We tried reinstalling the image but with no success

zhiquanyeo, 2021-02-04:
Does the issue persist across different SD cards?

julian-steiner, 2021-02-04:
We solved the issue by changing to a different raspberry pi, now a model 4 and it worked.

closed 2021-02-12 by PeterJohnson