wpilibsuite/WPILibPi #213: Enhancement - force switched (virtual) camera size and frame rate

jsimpso81, 2022-01-26:
The switched camera feature is really nice.
Consider the following enhancement(s).
 -- force the output frame rate, through a web dashboard setting.  This would allow streaming of cameras used for high rate vision processing to be streamed to the dashboard at a lower rate.
-- Resample the physical camera images at a fixed output size to allow switching between cameras using different pixel sizes.  Allow configuration of the output frame size on the dashboard

PeterJohnson, 2022-01-26:
Most (maybe all?) FRC dashboards allow setting of both resolution and FPS, as those are passed as GET parameters in the HTTP request for the stream.

GandalfTheGrey17, 2022-01-29:
Thanks.  Does this apply to the "virtual" switched camera too.   If we have a physical camera set to 30 fps, 640x480 pixels (and used for vision processing) can the switched camera send data to the dashboard at 10 fps, 320 x 240 pixels (for driver viewing)?  I've looked for a setting like this in the past, but didn't find.

jsimpso81, 2022-01-29:
Thanks. Does this apply to the "virtual" switched camera too. If we have a physical camera set to 30 fps, 640x480 pixels (and used for vision processing) can the switched camera send data to the dashboard at 10 fps, 320 x 240 pixels (for driver viewing)? I've looked for a setting like this in the past, but didn't find.

PeterJohnson, 2022-02-01:
Yes.  Per-stream settings (via GET parameters) can be applied to any stream including those coming from a switched virtual camera.  There will just be a cpu cost to rescale the image.

closed 2022-02-11 by jsimpso81