wpilibsuite/WPILibPi #214: will you guys going to update the image to the 2022 library ?

Ramtech59Sensei, 2022-01-27:


PeterJohnson, 2022-01-27:
As a lot of people use Python for image processing, we've been waiting for the 2022 version of RobotPy to stabilize before finishing the update (the rest of the update is ready to go).  Potentially we'll do an updated release this weekend.

PeterJohnson, 2022-02-12:
It’s basically ready to go, I just haven’t had a chance to test it.  Currently planning for a release this weekend now that I have my Pi up and running again.

3 USB cameras should work unless you’re running into USB bandwidth limits; I’ll test it again.  Do any errors appear in the console?

brhea, 2022-02-12:
W No console errors x When opening stream. It just doesn’t open the image or camera settings. I may try with lower frame rate.  We’re running low res 320x240 but 100 frames per sec.

closed 2022-02-13 by PeterJohnson