wpilibsuite/WPILibPi #215: Update base image to Raspi OS (Legacy)?

randomgrace, 2022-01-29:
Just installed from the current image.

An attempt to do an 'apt update' fails with:
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Ign:2 https://packages.cloud.google.com/apt coral-edgetpu-stable InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Err:4 https://packages.cloud.google.com/apt coral-edgetpu-stable Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP: 2607:f8b0:400a:805::200e 443]
Reading package lists... Done
E: Release file for http://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 245d 14h 55min 42s). Updates for this repository will not be applied.
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 246d 7h 34min 22s). Updates for this repository will not be applied.
E: The repository 'https://packages.cloud.google.com/apt coral-edgetpu-stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I can only assume that this is because buster is no longer supported directly.  If I am mistaken, I'd appreciate being corrected.

PeterJohnson, 2022-01-29:
That sounds more like your local time on the Pi isn't synchronized ("is not valid yet, invalid for another 245d").

randomgrace, 2022-01-29:
Got it in one.  I just now discovered that, and should have sussed it from that info.

Thanks.

closed 2022-01-29 by randomgrace