wpilibsuite/WPILibPi #219: Ability to be a network tables client for other than a team address

fovea1959, 2022-02-17:
Would it be possible to change the web UI to allow ip addresses and hostnames in addition to a team number if the rPi is set up as a network tables client? It would facilitate off-robot testing.

I am aware of being able to use the NetworkTables server on the rPi, but in this case, we were testing for some weirdness / race conditions in accessing an off-rPi NetworkTables instance.

PeterJohnson, 2022-02-17:
Duplicate of #140.  Agreed this would be a nice feature, it's just not been implemented yet.

zaaj, 2022-04-08:
One work-around on Windows hosts doing the off-robot testing is to use an ipv4 "Alternate Configuration" under the "Internet Protocol Version 4 (TCP/IPv4) control panel.  This is what Windows will use when it cannot reach a DHCP server.  Normally it defaults to "Automatic Private IP Address" in the APIPA range of 169.254.0.0/24, but you can put in a static IP address of 10.TE.AM.2 (normally used by the robot).  This is what I've done to get the pi's network tables client to send data to the OutlineViewer's server for testing our team's version of Peter's [rpi-colorsensor](https://github.com/first95/FRC2022/tree/main/piColorSensor)