wpilibsuite/WPILibPi #222: did the 2022 update change the romi's motor channels?

SammyPants37, 2022-03-21:
I just updated the romi image to 2022 and now the romi won't drive is this because of the update or some other issue on my end

jpokornyiii, 2022-03-23:
Can you confirm you can reach the romi via the webapi? Try http://wpilibpi.local or http://10.0.0.2. Re-imaging the Romi will reset the wireless configuration on the pi which could cause a mismatch with your Java code build configuration if you customized it.

SammyPants37, 2022-03-23:
I can reach http://10.0.0.2/ and this is a fresh image (I made this issue just a little while after I imaged it)

SammyPants37, 2022-03-23:
I just tested the romi reference code and that worked fine

jpokornyiii, 2022-03-23:
Can you post a link to the robot code that doesn't work? I'll try it out as well.

SammyPants37, 2022-03-23:
Yeah I can put it on GitHub at some point

jpokornyiii, 2022-03-24:
Can you check if you see "HALSimWS: WebSocket Connected" in the output as well? There were some changes between WPILib 2021 and WPILIB 2022 and the builld.gradle file. It may not be connecting at all.

SammyPants37, 2022-03-25:
here is my code that isn't working
https://github.com/SammyPants37/romiWork-2022

jpokornyiii, 2022-03-25:
I'll test this out when I get home tonight but my gut felling is that you are missing this in your build.gradle file:

wpi.sim.envVar("HALSIMWS_HOST", "10.0.0.2")
wpi.sim.addWebsocketsClient().defaultEnabled = true

This was coded differently in the past  and if you imported the project to FRC VS Code 2022 you would have to add them back.

jpokornyiii, 2022-03-25:
Change above fixed it when I tested it out. Please confirm this fixes the issue.

jpokornyiii, 2022-03-31:
Created above issue to follow up on the importing of romi projects to new VS Code versions. Will close this out. If this problem continues please create a new issue.

closed 2022-03-31 by jpokornyiii