wpilibsuite/WPILibPi #224: Initial Update of Romi Firmware Fails

OhHey-ItsDave, 2022-04-22:
Just received a "Romi for FRC" kit and RasPI 3B+, both new out of the box, and I am trying to do the initial setup following the WPI docs instructions. I have downloaded the latest WPILibPI image (v2022.1.1-Romi) and flashed it to a 16GB microSD card for the PI (the instructions say 8GB at minimum, 16 is the smallest I could find in a store). That worked with no issue, I can connect to the PI's webpage (wpilibpi.local) and can change to writable to specify a new SSID and password as the instructions show, but trying to update the 32U4 board's firmware always fails.

I can follow the WPI docs instructions for ["Imaging your Romi"](https://docs.wpilib.org/en/stable/docs/romi-robot/imaging-romi.html) up to the point of the Firmware Update for the 32U4: webpage is writeable, cable connecting the PI and 32U4 is in place, etc. but this is the output I get from the Firmware Update section of the PI's webpage:
```
Beginning binary upload to Romi ...
Resetting Romi to boot mode (should see quickly flashing yellow LED

Running imaging tool

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600

avrdude: ser_open(): can't open device "/dev/ttyACM0": No such file or directory

avrdude done.  Thank you.
```

I have searched this error on Google and Chief Delphi and found a few posts from about a year ago that recommended re-flashing the PI image with Etcher, checking to ensure the USB-to-microUSB cable that came with the Romi kit is a data cable and not a power-only cable, and ensuring that the batteries are fully charged. I have done all of these and still have had no luck.

Advice or direction would be appreciated, thank you!

PeterJohnson, 2022-04-22:
Give the last 2021 image a try.  It’s possible we accidentally broke something in the 2022 release.  Otherwise the most common cause is a USB power only cable; it seems like the vast majority of these cables are power only.

OhHey-ItsDave, 2022-04-22:
Tried again with the v2021.3.1-Romi image, upload failed the first time but worked the second time I did it a few seconds later. I don't know if comparing the output from the 2022 image to this one will help at all in debugging the new image but I'll include it here in case it is helpful to you. 

```
Beginning binary upload to Romi ...
Resetting Romi to boot mode (should see quickly flashing yellow LED

Running imaging tool

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
avrdude: ser_open(): can't open device "/dev/ttyACM0": No such file or directory

avrdude done.  Thank you.

Beginning binary upload to Romi ...
Resetting Romi to boot mode (should see quickly flashing yellow LED

Running imaging tool

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600

         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00

           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .

Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.

Programmer supports auto addr increment.

Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:

    Device code: 0x44

avrdude: devcode selected: 0x44

avrdude: AVR device initialized and ready to accept instructions

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file "/home/pi/.nvm/versions/node/v14.15.0/bin/../lib/node_modules/@wpilib/wpilib-ws-robot-romi/firmware/.pio/build/a-star32U4/firmware.hex"

avrdude: writing flash (11090 bytes):

avrdude: 11090 bytes of flash written
avrdude: verifying flash memory against /home/pi/.nvm/versions/node/v14.15.0/bin/../lib/node_modules/@wpilib/wpilib-ws-robot-romi/firmware/.pio/build/a-star32U4/firmware.hex:
avrdude: load data flash data from input file /home/pi/.nvm/versions/node/v14.15.0/bin/../lib/node_modules/@wpilib/wpilib-ws-robot-romi/firmware/.pio/build/a-star32U4/firmware.hex:

avrdude: input file /home/pi/.nvm/versions/node/v14.15.0/bin/../lib/node_modules/@wpilib/wpilib-ws-robot-romi/firmware/.pio/build/a-star32U4/firmware.hex contains 11090 bytes
avrdude: reading on-chip flash data:

avrdude: verifying ...

avrdude: 11090 bytes of flash verified

avrdude done.  Thank you.
```

Thanks for the help! 
Now I'll see if the 2021 Romi can talk to the 2022 VS Code.

OhHey-ItsDave, 2022-04-22:
Everything with VS Code is up and running, so I'm all set. Thank again for the help!

closed 2022-05-15 by OhHey-ItsDave