wpilibsuite/WPILibPi #226: Upgrading to bridge mode fails

jpokornyiii, 2022-11-26:
Occasionally, if I switch from AP to bridge mode via the web UI, the wpa_supplicant file fails to fully write my information and therefore I cannot start up in bridge mode. I checked wpa_supplicant_wpilibpi.conf and here is what was written:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

###### BELOW THIS LINE EDITED BY RPICONFIGSERVER ######
network={
    ssid="My Network"
    p
    
 workaround is to manually update this file to look like this:
  
 ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

###### BELOW THIS LINE EDITED BY RPICONFIGSERVER ######
network={
    ssid="My Network"
    psk-="MyPassword"
}

Note, this seems to happen randomly as if there is a race condition upon Save.

PeterJohnson, 2022-11-26:
Yeah, it looks like the file is only getting partially written.  There's nothing in the code that could cause that, so my guess is this is some kind of filesystem corruption happening due to a hard power-off following the change.  Right now we restart various services in place, but maybe a better approach is just to have it reboot following this kind of change?

veysby, 2023-01-19:
Constantly hitting this issue after upgrading to v2023.1.1

Steps to reproduce:
1. Write new image to SSD and start Romi
2. Connect to `WPILibPi-<number>` network and Navigate to [wpilibpi.local](http://wpilibpi.local) / Network Settings
3. Switch to writable mode and change Wifi Mode to Bridge
4. Provide SSID and WPA2 Passphrase and hit Save

First thing that happens on UI - SSID and WPA2 Passphrase becomes empty. After restart (I've tried both - hardware and using System Status / Restart System) Romi becomes unavailable. The only way to bring it back is like was suggested in <https://github.com/wpilibsuite/WPILibPi/issues/226#issue-1465192817> is to manually edit `wpa_supplicant_wpilibpi.conf` (in my case this file always empty)

seliger, 2023-12-27:
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev update_config=1 country=US
> 
> ###### BELOW THIS LINE EDITED BY RPICONFIGSERVER
> network={ ssid="My Network" psk-="MyPassword" }

Having this same exact problem consistently.

paulsp, 2024-08-02:
This issue also exists in v2023.2.1.

As noted above, a workaround is to add the following to /boot/wpa_supplicant_wpilibpi.conf.  Notice the key for the password is `psk` not `psk-` as in earlier posts.
``` 
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
ssid="mySSID"
psk="myPassword"
}
```