wpilibsuite/WPILibPi #227: Using older releases with 2023 RoboRio image

brhea, 2023-01-11:
Last year our team used the 2020 image of WPILibPi for vision because last year's release of WPILibPi was a bit late and everything worked with that image and the network table server on the roboRIO. 

I believe there were fairly significant changes to ntcore this year. Will older WPILibPi releases work with this year's release of network table server on the roboRIO? Any expected problems with this?

I have tried this on a prototype bot with all software other than WPILibPi and I do see values from the rPI updating.

PeterJohnson, 2023-01-11:
Yes, the 2023 ntcore NT4 server on the Rio is backwards compatible with NT3 clients like prior year WPILibPi images.

I am working on an updated Pi image as well, but it’s not going to be a required upgrade.

brhea, 2023-01-11:
Thanks Peter. We'll upgrade when new version becomes available. We'll test implementation of April tags when the new image is released.

redsil, 2023-01-14:
network tables is quite different in 2023, not sure the 2022 pi image could talk to it?  
Separately, would it be possible to include the socketcan drivers in your image for the RS485 CAN HAT?

auscompgeek, 2023-01-14:
> network tables is quite different in 2023, not sure the 2022 pi image could talk to it?

See:

> Yes, the 2023 ntcore NT4 server on the Rio is backwards compatible with NT3 clients like prior year WPILibPi images.

2023 ntcore supports both the NT3 and NT4 protocols.

closed 2023-01-23 by PeterJohnson