wpilibsuite/WPILibPi #230: Pi Camera Module 3 support

Krylez, 2023-02-18:
I'm having difficulty getting the Pi Camera Module 3 up and running. After reading a thread on [Chief Delphi](https://www.chiefdelphi.com/t/using-camera-module-with-raspberry-pi/424527), I added this to config.txt:
```
[all]
dtoverlay=imx708
```
After rebooting, there's no camera stream and `dmesg` shows:
```
[   16.522091] unicam fe801000.csi: Wrong metadata width/height/code 16384x1 00007002 (remote pad set to 28800x1 00007002)
[   16.522136] unicam fe801000.csi: Failed to start media pipeline: -22
[   31.716613] cam-dummy-reg: disabling
```
Just to confirm, the camera shows up as connected on the dashboard and I've  set the pixel format to YUYV and tried all of the below resolutions, which are [documented supported resolutions](https://www.raspberrypi.com/documentation/accessories/camera.html#hardware-specification):
* 2304 × 1296p56
* 2304 × 1296p30
* 1536 × 864p120

Using `damera_auto_detect=0` gets rid of the `dmesg` error, but there's still no camera stream.

It also makes no difference whether `media-controller=0` is there or not.

Any tips?

Krylez, 2023-02-18:
Thanks for replying so quickly. Just to confirm, I'll need to patch the changes from that commit to the source for the 2023.1.1 release and then build the OS?

Krylez, 2023-02-18:
So I tried carefully following along with the [build instructions for WPILibPi](https://github.com/wpilibsuite/WPILibPi), but I've had no success. I don't think I can devote the time to figuring that out on top of the other things we're working on. Is that the only way to fix this? Is the next release likely to be in a few months?

PeterJohnson, 2023-02-20:
Next release is imminent (uploading now).

PeterJohnson, 2023-02-20:
2023.2.1 is now released.  Please reopen this issue if you continue to have problems.

closed 2023-02-20 by PeterJohnson