wpilibsuite/WPILibPi #231: cpp-MulitCameraServer.zip contains libraries only built for 32-bit

S1ink, 2023-02-20:
It looks like the attached mulitcameraserver example for C++ in the latest [release](https://github.com/wpilibsuite/WPILibPi/releases/tag/v2023.2.1) contains 32-bit libraries instead of 64-bit ones like the last release. I have an automatic update script that pulls from the example project and extracts the libs/headers so that I can cross-compile for the pi on my windows machine, and now nothing can link properly because the libs are not 64-bit.

Heres the current proof:
1. All the libraries are smaller (and the entire .zip is smaller) than the last release.
2. I decompiled libapriltag.so and here are the comparative results:

v2023.1.1:
~~~
000000000002d770 <tagStandard52h13_destroy>:
   2d770:       a9be7bfd        stp     x29, x30, [sp, #-32]!
   2d774:       910003fd        mov     x29, sp
   2d778:       f9000bf3        str     x19, [sp, #16]
   2d77c:       aa0003f3        mov     x19, x0
   2d780:       f9401000        ldr     x0, [x0, #32]
   2d784:       97ff6dff        bl      8f80 <free@plt>
   2d788:       f9401660        ldr     x0, [x19, #40]
   2d78c:       97ff6dfd        bl      8f80 <free@plt>
   2d790:       f9401e60        ldr     x0, [x19, #56]
   2d794:       97ff6dfb        bl      8f80 <free@plt>
   2d798:       aa1303e0        mov     x0, x19
   2d79c:       f9400bf3        ldr     x19, [sp, #16]
   2d7a0:       a8c27bfd        ldp     x29, x30, [sp], #32
   2d7a4:       17ff6df7        b       8f80 <free@plt>
~~~
v2023.2.1:
~~~
0001ef64 <tagStandard52h13_destroy>:
   1ef64:       b510            push    {r4, lr}
   1ef66:       4604            mov     r4, r0
   1ef68:       6980            ldr     r0, [r0, #24]
   1ef6a:       f7e7 ee8a       blx     6c80 <free@plt+0x4>
   1ef6e:       69e0            ldr     r0, [r4, #28]
   1ef70:       f7e7 ee86       blx     6c80 <free@plt+0x4>
   1ef74:       6a60            ldr     r0, [r4, #36]   ; 0x24
   1ef76:       f7e7 ee84       blx     6c80 <free@plt+0x4>
   1ef7a:       4620            mov     r0, r4
   1ef7c:       e8bd 4010       ldmia.w sp!, {r4, lr}
   1ef80:       f7e7 be7c       b.w     6c7c <free@plt>
~~~
(If it isn't already obvious, 64-bit arm assembly does not have push/pop and uses r-registers vs x-registers)

PeterJohnson, 2023-02-20:
Ugh yeah I forgot it included the binaries.  I’ll need to separate that in the build process so I have two artifacts to upload.  In the meantime you can download the correct one from the Pi itself.

S1ink, 2023-02-20:
Ok thanks for such a fast reply! I'm sure I can find the libs after a bit of searching but would you happen to know the exact directory where they would be?

PeterJohnson, 2023-02-20:
There’s a zip file that is available through the vision page of the web dashboard and is also in /home/pi/examples I think

S1ink, 2023-02-20:
Oh yes, totally forgot about that. Thanks again!

PeterJohnson, 2023-02-21:
Added 64-bit version of the .zip to the release page.

closed 2023-02-21 by PeterJohnson